lemariva/micropython-camera-driver #1: Fails to init camera

mmphego, 2020-06-24:
Hi,
I just flashed my ESP32-CAM with your custom firmware and it fails to initialise the camera, with the message display below.
Please help.

[image]

Thanks in advance.

mmphego, 2020-06-24:
When I try to re-init the esp reboots.

[image]

lemariva, 2020-06-25:
Just ignore the error (GPIO isr service already installed), it is related with this [PR](https://github.com/espressif/esp32-camera/pull/145). The camera works without any problem.
If you try to initiate the camera twice, the board reboots. But as I said, initiate the camera once and ignore the error.

lemariva, 2020-06-25:
No, I used the code to get still images (timelapse), but you can check this [fork](https://github.com/krayon/upyesp32cam/commit/8b63edec50dca9416bb4b2b75207ac53788c597a), may be it helps. You need to adapt it to the new driver.

mmphego, 2020-06-25:
Thanks and keep those blog post coming.
I have bookmarked your page.

closed 2020-06-25 by mmphego

lemariva, 2020-06-27:
FYI, the streaming mode works. I added it to the master branch. You need to set a get parameter `stream=true` and you get a MPEG streaming. I've just updated the firmware to make it compatible to the application.

mmphego, 2020-06-28:
> FYI, the streaming mode works. I added it to the master branch. You need to set a get parameter `stream=true` and you get a MPEG streaming. I've just updated the firmware to make it compatible to the application.

How many fps on average?

lemariva, 2020-06-28:
very low: 2 fps

mmphego, 2020-07-10:
I got the same results, dissapointed. Had to end up using Arduino.
Thanks Mauro