lemariva/micropython-camera-driver #2: Invalid board on make

jcabrera96, 2020-07-16:
After following all steps on both tutorials I can't compile the firmware (DIY) ` with make BOARD=GENERIC_CAM PYTHON=python3 MICROPY_PY_BTREE=0 -j`.

The command returns 'Makefile:10: *** Invalid BOARD specified: boards/GENERIC_CAM.  Stop.' as there's no folder named GENERIC_CAM at boards/, neither at the repository.

I have cloned micropython repository, installed ESP-IDF framework, cloned esp32-cam repository into ~/esp/esp-idf/components, cloned this repository into separate folder and copied its content into %micropython repo path%/ports/esp32.

May I be missing something?

If I compile it using GENERIC_SPIRAM it enters eternal reboot loop with the folowing info:

```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40097afc  PS      : 0x00060d30  A0      : 0x800d2b77  A1      : 0x3ffcf5a0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00007d11  A5      : 0x00000000  
A6      : 0x3ffc3b94  A7      : 0x000007d1  A8      : 0x800fbce9  A9      : 0x3ffcf570  
A10     : 0x3ffc2d68  A11     : 0x3ffba184  A12     : 0x3ffcf590  A13     : 0x0000000c  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40097afc  LEND    : 0x40097b07  LCOUNT  : 0x000007d0  

ELF file SHA256: 0000000000000000

Backtrace: 0x40097af9:0x3ffcf5a0 0x400d2b74:0x3ffcf5b0 0x400f79f1:0x3ffcf5d0 0x40093e8d:0x3ffcf610

Rebooting...
```

Any advice is welcome.

Thanks in advance.

lemariva, 2020-07-16:
Hi, thanks for the issue. 
Sorry, I've commited the wrong BOARD folder. I've just changed it. Can you check if it works right now? Sorry again!

jcabrera96, 2020-07-16:
Hello again.

It's working perfectly with the new folder!

Thanks for the quick answer and your hard job!

closed 2020-07-16 by jcabrera96