lemariva/micropython-camera-driver #6: Detected camera not supported

kielnino, 2020-08-30:
Hi, thank you for your effort. 
Unfortunately I get the following error, when I try to init the camera:
```
>>> camera.init(0)
I (461978) sccb: pin_sda 26 pin_scl 27

E (461988) i2c: i2c driver install error
I (461988) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (462038) sccb: SCCB_Write Failed addr:0x30, reg:0xff, data:0x01, ret:-1
E (462048) sccb: SCCB_Write Failed addr:0x30, reg:0x12, data:0x80, ret:-1
E (462098) camera: Detected camera not supported.
E (462098) camera: Camera probe failed with error 0x20004
E (462098) camera: Camera Init Failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed
```
I have an ESP-32S Cam Breakout which is labeled "DIY MORE". I noticed, that there are different pinouts on the market (like listed [here](https://github.com/bitluni/ESP32CamTimeLapse/blob/master/TimeLapse/camera_pins.h)). Is the error maybe related to a wrong call on the `init`-function? Do I need to adjust the call and define the pinout somehow?

kielnino, 2020-08-30:
However somtimes I get a different error:
```
>>> camera.init(0)
I (14988) sccb: pin_sda 26 pin_scl 27

I (14998) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (15148) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15148) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15148) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15158) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15168) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15178) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15188) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15198) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15208) gpio: GPIO[25]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15218) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15228) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (15238) camera: Allocating 1 frame buffers (234 KB total)
I (15268) camera: Allocating 234 KB frame buffer in OnBoard RAM
E (15308) gpio: gpio_install_isr_service(411): GPIO isr service already installed
W (15308) camera: gpio_install_isr_service already installed
E (16358) camera: Timeout waiting for VSYNC
E (16368) camera: Camera init failed with error 0x20003
E (16368) camera: Camera Init Failed
```
Could it be related to an insufficient power supply (USB port on PC)?

lemariva, 2020-08-30:
You need to configure the camera pins. The default option is for the [ESP32-CAM](https://bit.ly/2Ndn8tN). You can check the M5Camera configuration:
```
# M5Camera (Version B) - https://bit.ly/317Xb74
camera.init(0, d0=32, d1=35, d2=34, d3=5, d4=39, d5=18, d6=36, d7=19,
            format=camera.JPEG, xclk_freq=camera.XCLK_10MHz,
            href=26, vsync=25, reset=15, sioc=23, siod=22, xclk=27, pclk=21)   #M5CAMERA
```
and you need to change the `d0` to `d7`, `href`, `vsync`, `reset`, `sioc`, `siod`, `xclk`, and `pclk` according to the board datasheet, or the h-file that you [posted](https://github.com/bitluni/ESP32CamTimeLapse/blob/master/TimeLapse/camera_pins.h).

kielnino, 2020-08-31:
I think my module is faulty. After restarting the ESP several times I can successfully initialize the camera.

closed 2020-08-31 by kielnino