lemariva/micropython-camera-driver #8: GPIO isr service already installed

aguaviva, 2020-09-25:
I noticed this error 
```
E (2498) gpio: gpio_install_isr_service(411): GPIO isr service already installed
W (2498) camera: gpio_install_isr_service already installed
```
The error one can only see it when an app is launched from boot.py, if you launch the script from Thonny you wont see it

The problem (and the solution) is explained here
https://github.com/espressif/esp32-camera/pull/145

P.D: Time to go to bed, thanks for this wonderful firmware!

aguaviva, 2020-10-03:
This issue could be related to this

https://github.com/micropython/micropython/issues/6512

lemariva, 2020-10-18:
I updated the driver to the last commit (48a83ed97be1642e71f148333290c86f480b3cbb) which included that PR, but the camera stops working. I had to undo that :(

stanisljevic, 2021-03-29:
`E (69354) gpio: gpio_install_isr_service(411): GPIO isr service already installed`
I have the same problem, did you manage to fix this, thank you

lemariva, 2021-04-05:
Hi, as I said some time ago, the error and the warning do not stop the camera from working. Just ignore it.

aguaviva, 2021-04-05:
The problem comes when you try to implement a timelapse app, it will only wake up from sleep a number of times until it finally hangs forever.

lemariva, 2022-01-08:
Using the new driver [221d24d](https://github.com/espressif/esp32-camera/commit/221d24da1901b6aee8df6c1a1160ad02faa685b5), I still got same warning. Can you check if the ESP32 still hangs forever with the latest version that I push it today? Thanks!

lemariva, 2022-02-02:
closing because of no activity. As I said, check the latest driver, and re-open if it's needed.

closed 2022-02-02 by lemariva

alirezadigi, 2022-03-27:
hi dear @lemariva 
i self compiled firmware
and i have this problem.
the app hungs ...

alirezadigi, 2022-03-27:
it seems that app is capturing to sd card
but i have no access via browser