lemariva/micropython-camera-driver #9: `ujson` fails to install

stggh, 2020-09-26:
A magnificent project!

I can capture static images, but the streaming part fails due to a failure to install `ujson`, as per your tutorial [#MicroPython: Taking photos with an ESP32-CAM](https://lemariva.com/blog/2019/09/micropython-how-about-taking-photo-esp32)

[image]

```python
MicroPython v1.13-4-g3ff707927-dirty on 2020-09-06; ESP32 CAM with ESP32
Type "help()" for more information.
>>> 
>>> import upip
>>> upip.install('ujson')
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Error installing 'ujson': , packages may be partially installed
```
Using firmware `micropython_3ff7079_esp32_idf4.x_ble_camera.bin`

stggh, 2020-09-26:
Hmm, ujson appears to installed, via the firmware:
```python
[Rpi0W:.+stg121/Software]$ !rsh
rshell -p /dev/serial0
Using buffer-size of 32
Connecting to /dev/serial0 (buffer-size 32)...
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /boot.py/ /buf.txt/ /lib/
Setting time ... Sep 26, 2020 18:07:00
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/home/stg121/Software> repl
Entering REPL. Use Control-X to exit.
>
MicroPython v1.13-4-g3ff707927-dirty on 2020-09-06; ESP32 CAM with ESP32
Type "help()" for more information.
>>> 
>>> import ujson
>>> ujson.
__class__       __name__        dump            dumps
load            loads
```
my lack of streaming is elsewhere.  Closing this issue.

closed 2020-09-26 by stggh

patniemeyer, 2021-04-04:
Is the error actually due to the package already being installed or is it an unrelated error having to do with upip being stale? (I came here for the same reason and am pleased to see the resolution is "do nothing" but curious if that is the issue).