lemariva/micropython-camera-driver #10: Issue building cam firmware

keenankyle, 2020-10-12:
Good day, 

I've been atttempting to build the latest micropython firmware with the camera driver included. I was successful in the past on version 1.12 however now I am hitting the following error when i run the make command:

make BOARD=GENERIC_CAM PYTHON=python3 MICROPY_PY_BTREE=0 -j
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Building with ESP IDF v4
GEN build-GENERIC_CAM/sdkconfig.h
make: *** No rule to make target 'nimble.c', needed by 'build-GENERIC_CAM/genhdr/qstr.i.last'.  Stop.
make: *** Waiting for unfinished jobs....

Can anyone assist, I don't know what I am doing wrong.

lemariva, 2020-10-18:
The problem is that you've cloned the latest commit of the MicroPython repository and replaced the original files with the files in this repository. I provided the commit hash to avoid this type of issue. MicroPython changes a lot.
In this case, the `nimble.c` was renamed to `mpnimbleport.c`, therefore the compiler didn't find it. I've just updated the files, I've also provided the hash that I used. If you want to start from scratch, I'd recommend you to follow the steps and modify the original files according to the instructions provided.

closed 2020-10-18 by lemariva

keenankyle, 2020-10-28:
Thanks so much, the build was now successful.