lemariva/micropython-camera-driver #12: Init Failed

citakis, 2020-10-26:
I have a Chinese ESP32CAM module and I had a lot of Camera Init Faled errors. Camera was initialized correctly and working one in four times, and usually if init failed the only solution was to power off completely not soft reboot otherwise it would never pass the init.

I tried many solutions including external power supply instead of USB, changed the camera module with another one and I always had the same behaviour. The only solution I found was to declare the pins from the beginning.
For example for my camera seems to be working the following setup
camera.init(0, d0=5, d1=18, d2=19, d3=21, d4=36, d5=39, d6=34, d7=35, href=23, vsync=25, sioc=27, siod=26, xclk=0, pclk=22, pwdn=32)
The pinout was taken from the espressif arduino web camera example which was working without errors
Now it works almost flawlessly.
I don't know if it is something  wrong with the firmware or the 2640 driver but I thought to post my solution in case it helps someone else.

jansenarruda, 2020-10-26:
Hi,
I have a TTGO camera version 1.6.2

The camera works in arduino flawless, but when I try to camera.init, my micropython reboot´s 

the parameters for my init was: camera.init(0, d0=34, d1=13, d2=14, d3=35, d4=39, d5=38, d6=37, d7=36,href=27, vsync=5, sioc=23, siod=18, xclk=4, pclk=25,   xclk_freq=camera.XCLK_20MHz,reset=-1, pwdn=-1, )

In arduino I use the same PINs numbers. I try the firmware in github and try to build, same behavior.

I try to config the PINs directly in modcamera.h, but same error.

thank you !!!

jansenarruda, 2020-10-27:
Hi

I managed to solve the reboot when init camera (my bad, my configs was in bad shape and don't create a SPIRAM firmware). When I managed to compile a SPIRAM firmware the init work´s (give me 'E (17690) gpio: gpio_install_isr_service(411): GPIO isr service already installed' but work´s). Now I had another issue... try to use ssd1306 reboot the board...

Thank´s

lemariva, 2020-11-07:
Try to use the soft SPI (-1) for the SSD1306.

jansenarruda, 2020-11-09:
I´ll try!!! I´ll build again with the new version and test

keenankyle, 2020-11-15:
I'm having the same issues when trying to call the init cam method. I've tried rebuilding the firmware, it made little difference. I am calling a reboot in my script if the init fails, this seem to help at times, as after about 3 or so attempts the camera initializes succesfully. This however is not very reliable, as the device will eventually get stuck in a boot loop. Could use some suggestions on how to fix this

skeyvin, 2022-01-01:
> Try to use the soft SPI (-1) for the SSD1306.

hi, how?

lemariva, 2022-01-08:
I should close the issue before! 
@skeyvin: soft or hardware SPI versions are now different instructions. Check these links: 
* https://docs.micropython.org/en/latest/esp32/quickref.html#software-spi-bus
* https://docs.micropython.org/en/latest/esp32/quickref.html#hardware-spi-bus

closed 2022-01-08 by lemariva