lemariva/micropython-camera-driver #13: I can not init camera

maysrp, 2020-11-13:
When I use JPG initialization, there is no problem, and I use rgb565 initialization will report an error。

camera.init(0, format=camera.RGB565)
E (118749) camera: Allocating 3750 KB frame buffer Failed
E (118749) camera: Failed to allocate frame buffer
E (118749) camera: Camera init failed with error 0x101
E (118759) camera: Camera Init Failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed

maysrp, 2020-11-13:
ESP32-cam

lemariva, 2021-01-03:
yes, the problem is that the image it too big for the RAM. Check that with an smaller frame size. But, I think the format generates buffers which are too big for the RAM.

closed 2021-04-05 by lemariva