lemariva/micropython-camera-driver #14: camera init error in esp eye

cristianpb, 2020-12-08:
Hi,

I just flashed your firmware `micropython_b7883ce_esp32_idf4.x_ble_camera.bin` in a esp-eye. I tried to load the camera using the init method but I got the following error:

```
>>> import camera
>>> camera.init(0, format=camera.JPEG)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed
>>>
```

Do you know where could be the error ? Do I have to use other camera driver for this device ? 

Thanks,

lemariva, 2021-01-03:
Check the pin configuration inside the config.py file. You need to define that for the ESP-EYE camera. Keep me updated.

cristianpb, 2021-01-03:
I started working with other code and I realize about the pin configuration.

For example, this pin configuration for the m5stack and the esp-eye

```cpp
#elif defined(CAMERA_MODEL_ESP_EYE)
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    4
#define SIOD_GPIO_NUM    18
#define SIOC_GPIO_NUM    23

#define Y9_GPIO_NUM      36
#define Y8_GPIO_NUM      37
#define Y7_GPIO_NUM      38
#define Y6_GPIO_NUM      39
#define Y5_GPIO_NUM      35
#define Y4_GPIO_NUM      14
#define Y3_GPIO_NUM      13
#define Y2_GPIO_NUM      34
#define VSYNC_GPIO_NUM   5
#define HREF_GPIO_NUM    27
#define PCLK_GPIO_NUM    25

#elif defined(CAMERA_MODEL_M5STACK_PSRAM)
#define PWDN_GPIO_NUM     -1
#define RESET_GPIO_NUM    15
#define XCLK_GPIO_NUM     27
#define SIOD_GPIO_NUM     25
#define SIOC_GPIO_NUM     23

#define Y9_GPIO_NUM       19
#define Y8_GPIO_NUM       36
#define Y7_GPIO_NUM       18
#define Y6_GPIO_NUM       39
#define Y5_GPIO_NUM        5
#define Y4_GPIO_NUM       34
#define Y3_GPIO_NUM       35
#define Y2_GPIO_NUM       32
#define VSYNC_GPIO_NUM    22
#define HREF_GPIO_NUM     26
#define PCLK_GPIO_NUM     21
```

For the esp eye the configuration pins should be:

```python
camera.init(0, d0=34, d1=13, d2=14, d3=35, d4=39, d5=38, d6=37, d7=36,
            format=camera.JPEG, framesize=camera.FRAME_VGA, xclk_freq=camera.XCLK_10MHz,
            href=26, vsync=5, reset=-1, sioc=23, siod=18, xclk=4, pclk=25)   #ESP-EYE
```

I will try this in the esp eye and if it works I would be glad to make a PR if you are ok

lemariva, 2021-04-05:
Sure, Did it work?

lemariva, 2022-01-08:
no activity closing

closed 2022-01-08 by lemariva

alwayscommit, 2022-03-23:
Any solution for this problem?

wldeveloperapps, 2022-04-21:
Any solution?