lemariva/micropython-camera-driver #16: jpg images have missing data at the bottom (because of maximum file size)

lorenz-g, 2020-12-30:
Hello, thanks again for porting the esp32 cam driver to micropython. It works really well. 

I have noticed that for framesize 5, the maximum size of the pictures is 9600 bytes. 
However I sometimes get pictures that are larger and are cut at the bottom. 
The same is for framesize 3 with 5280 bytes. (see the two sample pictures below). 

Could you point me to the direction where you can set the maximum size for jpg images?
I will also try to figure it out myself and will update back here. 

I am using the `micropython_b7883ce_esp32_idf4.x_ble_camera.bin` firmware with the esp32 ai thinker camera module. 

[image]
[image]

lemariva, 2021-01-03:
RAM is limited and that's the problem. I could set a bigger buffer, but then you'll get "out of memory" errors on the application. If you want to try, you can modify the line:
```
mp_task_heap_size = 2 * 1024 * 1024; 
```
from the file `main.c`. Keep me updated, if you get better results.

lorenz-g, 2021-01-04:
Hi, thanks for the hint with the RAM. I will try it out. 

But it might be sth different, because with framesize 11, I get can get pictures that are fine and are 60kB large.
And it also happens sometimes from framesize 1 where the pictures is only 2kB (see the two images below). 

As a cross check I will also take similar pictures with the standard c code and check it it has a similar behaviour. 

[image]
[image]

lemariva, 2022-01-08:
I've updated the firmware, could you check that again?

lemariva, 2022-02-02:
@lorenz-g can you check the latest driver? I think that could be a problem with the HEAP/SPIRAM. Now MicroPython works better with that and shares the PSRAM with other IDF components.

lorenz-g, 2022-02-02:
Hi Mauro, I will check it with the latest driver but might need a week or two. Sorry.

lemariva, 2022-02-02:
No problem! Note the new parameter `fb_location`. If you are using the ESP32_CAM you can set it to `camera.PSRAM`.