lemariva/micropython-camera-driver #17: Camera probe failed with error 0x20001

pavelrevak, 2021-01-03:
on M5CAMERA I getting this error:
I have flashed this `b7883ce` firmware
```
>>> import camera
>>> camera.init(0, format=camera.JPEG)
E (279771) camera: Camera probe failed with error 0x20001
E (279771) camera: Camera Init Failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed
```
thanks for any advise

lemariva, 2021-01-03:
sorry, I didn't see that error.
Try to erase the flash, flash firmware again and check the pins setting.

pavelrevak, 2021-01-03:
thanks for quick answer, what pins settings did you think?
reflashed:
```
$ esptool.py --port /dev/tty.usbserial-01BB6216 erase_flash                                                                                    
esptool.py v3.0
Serial port /dev/tty.usbserial-01BB6216
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:07:07:b4
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.4s
Hard resetting via RTS pin...

$ esptool.py --port /dev/tty.usbserial-01BB6216 --baud 460800 write_flash -z 0x1000 ~/Downloads/micropython_b7883ce_esp32_idf4.x_ble_camera.bin
esptool.py v3.0
Serial port /dev/tty.usbserial-01BB6216
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:07:07:b4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 1567824 bytes to 982369...
Wrote 1567824 bytes (982369 compressed) at 0x00001000 in 24.0 seconds (effective 523.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

from restart:
```
rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5180
load:0x40078000,len:14032
load:0x40080400,len:3800
entry 0x4008063c
MicroPython v1.13-153-g9c0ac7e6a-dirty on 2020-11-07; Camera Module (i2s) with ESP32
Type "help()" for more information.
>>> import camera
>>> camera.init(0, format=camera.JPEG)
E (53240) camera: Camera probe failed with error 0x20001
E (53240) camera: Camera Init Failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed
```
camera module is this:
[image]

pavelrevak, 2021-01-03:
I see this, sorry, for M5 is different init..

```
>>> camera.init(0, d0=32, d1=35, d2=34, d3=5, d4=39, d5=18, d6=36, d7=19, format=camera.JPEG, framesize=camera.FRAME_VGA, xclk_freq=camera.XCLK_10MHz, href=26, vsync=25, reset=15, sioc=23, siod=22, xclk=27, pclk=21)
E (20933) gpio: gpio_install_isr_service(411): GPIO isr service already installed
True
```
is it OK ?

pavelrevak, 2021-01-03:
Ok, it is working as expected. thanks!

closed 2021-01-03 by pavelrevak