lemariva/micropython-camera-driver #19: Saturation/Brightness/Contrast don't seem to do anything

dracode, 2021-03-04:
Using the firmware you provide here (thanks!) and your uPyCam project completely unmodified, except for config.py:
```
app_config = {
    'camera': 'ESP32-CAM',  # camera -> 'ESP32-CAM' or 'M5CAMERA'
    'led': 4, # led -> 4: ESP32-CAM or 14: M5CAMERA
}
```

and of course my wifi parameters.

When I change the saturation, brightness, or contrast, nothing seems to change in the retrieved image.  Most notably is saturation, which according to the README should result in grayscale when set to -2: i continue to get full color.
All other options work as expected.

Any ideas?

lemariva, 2022-01-08:
I've updated the firmware, could you check that again?

lemariva, 2022-02-02:
closing because of no activity

closed 2022-02-02 by lemariva