lemariva/micropython-camera-driver #20: Port to CMake

TecDroiD, 2021-03-31:
Hi,
seems that micropython uses CMake lately. So your DIY-tutorial doesn't work anymore.
Will you port your firmware to CMake, too?

lorenz-g, 2021-03-31:
What a coincidence, I have just opened a PR where I documented the build process with Cmake :) 
https://github.com/lemariva/micropython-camera-driver/pulls 

Let me know if this worked for you.

TecDroiD, 2021-03-31:
Thank you! This worked- even if i didn't get a prompt via picocom yet. 
For an esp32 beginner (like me) it is hard to figure out which parts in the referenced compile tutorial one has to follow. Maybe you can just copy the neccessary text lines?

lorenz-g, 2021-04-03:
Hi, 
I am not an expert on compiling micropython either :) Might be best if you submit a PR yourself on how you think to improve the documentation. 

Regards,
Lorenz

uraich, 2021-09-20:
Since I get an error when the camera driver allocates memory for the frame buffer, I tried to re-create the firmware with exactly the versions of esp-idf, esp32-camera driver and micropython that are described in the git repository. 
When trying to check out version 2dded7c of esp-idf I get:
error: pathspec '2dded7c' did not match any file(s) known to git.

When using the new versions of these software elements, I get :

Initializing OV2640
OV2640 Attached
D (18883) ov2640: OV2640 Attached
I (18883) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
D (18893) camera: Doing SW reset of sensor
I (18983) cam_hal: buffer_size: 32768, half_buffer_size: 4096, node_buffer_size: 2048, node_cnt: 16, total_cnt: 93
I (18983) cam_hal: before heap_caps_calloc, frame_cnt: 1 

Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
Do you have any idea how I can create a version that works?
I also tried to flash firmware.bin from the repository and that works correctly. This means that I must make an error somewhere.

lemariva, 2022-01-08:
I got that error too. It seems, it is related with the PSRAM. I explained that in the new README.md. I wrote the solution too.

closed 2022-01-08 by lemariva