lemariva/micropython-camera-driver #23: SPIRAM unused

TecDroiD, 2021-04-06:
Hi again, i just found out that SPI RAM on ESP32-Cam is unused.
Is it possible to implement this? i'll try on my own but i'm not really convinced to have enough insight into mp :-(
[Here is a modified image (from 2018) with modifications that have been done..](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo)

lemariva, 2022-01-08:
Can you point the modifications, or send me a PR? The flags:
```
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
CONFIG_SPIRAM_USE_MALLOC=y
```
which are related with the PSRAM are set to `y` (yes) - check the sdkconfig.esp32cam file.

lemariva, 2022-01-31:
You're right! MicroPython takes over the entire SPIRAM region and uses it for the GC heap. The camera is using the internal RAM for the DMA buffer. If you want to change that, you need to modify the logic inside the `main.c` file. I tried that but without sucess.

lemariva, 2022-02-02:
Not anymore. You were right. But since [PR 8219](https://github.com/micropython/micropython/pull/8219) the camera can share the PSRAM with MicroPython. Check out the latest firmware and the camera option `fb_location=camera.PSRAM`. With that option, you can take photos with higher resolution.

closed 2022-02-02 by lemariva

TecDroiD, 2022-02-02:
thank you, i'll try