lemariva/micropython-camera-driver #29: HiLetgo ESP32 camera failed to init

keredson, 2021-08-24:
this one:  https://www.amazon.com/gp/product/B07X1W2PT3

```
>>> import camera
>>> camera.init(0, format=camera.JPEG)  
E (293124) camera: Detected camera not supported.
E (293124) camera: Camera probe failed with error 0x20004
E (293124) camera: Camera Init Failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed
```

keredson, 2021-08-24:
seems to have worked:
```
>>> import camera
>>> camera.init(0,
...   d0=5,
...   d1=14,
...   d2=4,
...   d3=15,
...   d4=18,
...   d5=23,
...   d6=36,
...   d7=39,
...   href=25,
...   vsync=27,
...   sioc=12,
...   siod=13,
...   xclk=32,
...   pclk=19,
...   xclk_freq=camera.XCLK_10MHz,
...   reset=-1,
...   pwdn=26,
... )
E (23905) gpio: gpio_install_isr_service(460): GPIO isr service already installed
True
>>> buf = camera.capture()
>>> len(buf)
84079
>>> 
```

aguaviva, 2021-11-27:
What is issue here? That you need extra paramenter to set the pins of the camera?

lemariva, 2022-01-08:
@aguaviva :+1:  @keredson Thanks for the board link.

closed 2022-01-08 by lemariva

keredson, 2022-01-08:
> What is issue here? That you need extra paramenter to set the pins of the camera?

Just helping document, since it seemed there's automated detection that didn't work for this board.