lemariva/micropython-camera-driver #32: camera driver shouldnt require making changes in micropython

aguaviva, 2021-11-27:
Hi Mauro,

I know you put this project a long time ago and you did a great job. 

There is a very interesting way to structure the driver so people don't need to make any manual changes to micropython, it will just pick the new module where ever it is.

A good example of that is this TDT driver:   https://github.com/russhughes/ili9342c_mpy

It takes no time to setup and compile, I hope you like it! :)

Cheers and thanks for this fantastic driver!!

lemariva, 2022-01-08:
Done, thanks for the link. 
* I couldn't integrate one modification: for the PSRAM related I couldn't find a solution. 
* I couldn't switch off the CMake flag for the DBTree library and I got some problems with it (thus, I modified some other files).

mocleiri, 2022-02-02:
@lemariva https://github.com/micropython/micropython/pull/8219 just got merged which removes the need to modify micropython for PSRAM changes.

As long as the sdk config sets **CONFIG_SPIRAM_USE_MALLOC=y** then micropython will now only claim 1/2 of the available PSRAM.  So with ESP32-CAM-MB its 4MB of PSRAM is split 2MB for the micropython heap and 2MB for esp32-camera or other idf components.

I included a modified version of your library(I expect to try and upstream the differences) in https://github.com/mocleiri/tensorflow-micropython-examples/ we are building https://github.com/mocleiri/tensorflow-micropython-examples/tree/main/boards/esp32/MICROLITE_SPIRAM_CAM without any DBTree changes.

Thank you for making this library as it enabled us to get the tensorflow lite for microcontrollers person detection example working.

lemariva, 2022-02-02:
@mocleiri thanks for posting this information. I've just updated the driver and works really nice. 
Now it's possible to take HD photos with the camera. I added an argument `fb_location` to enable the use to select between PSRAM or DRAM. Summarizing: 
* The DBTree modifications were a problem with my disk format. I cloned the repository inside my encrypted drive (extended FAT) and then moved it to a Ext4 because of the symbolic links (those doesn't work under FAT). But something was still there.
* The main.c file doesn't need any modification. 

I'll check out your library, looks very promising!

closed 2022-02-02 by lemariva