lemariva/micropython-camera-driver #35: Guru Meditation Error: Core  0 panic'ed (StoreProhibited)

ernitron, 2022-01-15:
I followed to the letter your instructions for recompiling micropython. I even tried this last update using the USER_C_MODULES. Compilation was OK. Used esp-idf.py 4.2. But I get Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Which seems different (LoadStoreAlignment) you mentioned. I tried different, devices, power configurations, ways to upload firmware all with same results.

Besides, I get exactly the same error when trying your latest firmware.bin so I suspect something different but I have no clue. 

Any hint will be appreciated as I am really stuck.

This is the output from serial (looks like panic message from ESP32 FreeRTOS but I cannot decode what it means):

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:4536
load:0x40078000,len:12788
load:0x40080400,len:4176
entry 0x40080680
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d9c7  PS      : 0x00060830  A0      : 0x8010609c  A1      : 0x3ffca1a0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00007d11  A5      : 0x00000000  
A6      : 0x3ffca1a0  A7      : 0x000007d1  A8      : 0x800f83d9  A9      : 0x3ffca170  
A10     : 0x3ffb5140  A11     : 0x3ffbadf8  A12     : 0x00000003  A13     : 0x00000000  
A14     : 0x00001000  A15     : 0x00000008  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0x4008d9c4:0x3ffca1a0 0x40106099:0x3ffca1b0 0x400f40d6:0x3ffca1d0

ELF file SHA256: 00705d4c6b010b7e`

ernitron, 2022-01-15:
I just got back to the original main.c without the fix suggested in your point 2 and it works now, meaning that the firmware.bin starts gracefully without exception to memory fault. Well, I am struggling now with camera.init(0) as I guess I have to figure out the right parameters for my camera but at least the problem related to addressing out of space seems solved. What do you think?

lemariva, 2022-01-15:
What board are you using right now? Both errors StoreProhibited/LoadStoreAlignment are related with the PSRAM and those lines in the `main.c` file. Thus, our boards may be different. It's good to know that the original `main.c` works with your board, but not with my ESP32-CAM and M5Stack. 

The camera.init(0) is only for the ESP32-CAM module. If the camera is not found, you get the error [0x105](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/error-codes.html).

ernitron, 2022-01-15:
Hi Marco, thanks for replying. 
As a matter of fact, the original main.c seems to mange the different kind of HW and in my case is working. 

I am using this ESP32-CAM like this one https://m.media-amazon.com/images/I/71LGsWic1pS._SL1500_.jpg

This is what esptool says:
```
micropython-camera-driver$ esptool.py chip_id
esptool.py v3.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:eb:d6:7b:84:b4
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 4c:eb:d6:7b:84:b4
Hard resetting via RTS pin...
```
I figure it out how to initialize camera and now is working

```
>>> import camera
>>> camera.init(0, format=camera.JPEG, framesize=5)
True
>>>
```
I will give a try to your uPyCam.  I just want to thank you for your great work.

lemariva, 2022-02-02:
Check out the new driver it should work better, and the camera can use the PSRAM (`fb_camera=camera.PSRAM`). Since [PR8219](https://github.com/micropython/micropython/pull/8219) MicroPython can share the PSRAM with other idf components.

closed 2022-02-02 by lemariva