lemariva/micropython-camera-driver #36: Error while trying to write files in the virutal file system.

ncotti, 2022-01-27:
Hi, I recently upgraded the firmware to the new version you provided, but i've encountered a strange bug.

My code looks like this:

```
with open("/log.txt" , "a+") as file:
    pass

camera.init(0, format=camera.JPEG)

for i in range(0,100):
    with open("/log.txt", 'a+') as log_file:
        print ("Test " + str(i))
        log_file.write("Test " + str(i) + "\n")
```

As you can see, after initializing the camera, I try to perform multiple write operations on a single file. However, after an undefined amount of these operations, the program execution blocks in the "write" statement and never leaves. I present to you some terminal output examples:

```
Test 0
Test 1
```

```
Test 0
Test 1
Test 2
```
After further examination, i have discovered that this issue only happens if the file, in this example "/log.txt`, existed before resetting the ESP32. What I mean is that doing this kind of operation works fine as long as the file is created in the same execution instance. However, if you happen to use the open mode with "a" (so it won't be created) and the file existed before powering on the ESP32, the writing operation will block.

lemariva, 2022-02-02:
Can you check out the new firmware? I think that could be a problem with the HEAP. Now it's better managed.

lemariva, 2022-02-03:
Thanks for reaching out! I close the issue then.

closed 2022-02-03 by lemariva