lemariva/micropython-camera-driver #37: Camera won't initialise

alport, 2022-02-02:
Using Thonny, I've successfully installed your latest binary: https://github.com/lemariva/micropython-camera-driver/blob/master/firmware/micropython_camera_feeeb5ea3_esp32_idf4_4.bin
and get the Micropython prompt,

but, I can't seem to get past the simplest test:
>>> import camera
>>> camera.init(0, format=camera.JPEG, fb_location=camera.PSRAM)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed

Any pointers please?
Thanks,

lemariva, 2022-02-03:
Hey! can you please flash the firmware using `esptool` and connect to the board using VSCode and the PyMakr? The firmware seems to be working by other users e.g. https://github.com/lemariva/micropython-camera-driver/issues/36#issuecomment-1029051970 . Thus, I think the IDE is the problem. You can follow the steps here: https://lemariva.com/blog/2022/01/micropython-upgraded-support-cameras-m5camera-esp32-cam-etc

aguaviva, 2022-02-12:
Same issue here with the AIThinker esp32-cam

>import camera
>camera.init(0, format=camera.JPEG, fb_location=camera.PSRAM)
>Traceback (most recent call last):
>  File "<stdin>", line 6, in <module>
> OSError: Camera Init Failed

I flashed your firmware this way:
> esptool.py --port /dev/ttyUSB0 erase_flash
> esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash  -z 0x1000 ~/Downloads/micropython_camera_feeeb5ea3_esp32_idf4_4.bin

With the one I compiled I got a crash in cam_hal.c
>    cam_obj->dma_buffer = NULL;
>    cam_obj->dma = NULL;
>
>    cam_obj->frames = (cam_frame_t *)heap_caps_calloc(1, cam_obj->frame_cnt * sizeof(cam_frame_t), MALLOC_CAP_DEFAULT);
>    CAM_CHECK(cam_obj->frames != NULL, "frames malloc failed", ESP_FAIL);
>
>    uint8_t dma_align = 0;

ABmamunHoneybee, 2022-08-24:
@aguaviva  did you get any solution? Please help me.
Same problem I faced. Please give a solution @lemariva

@lemariva, Also, I flash like @aguaviva and restore it again. but problems are the same.

jimrafox, 2023-06-08:
Hello, I know this is an old thread, but I have the same problem. After flashing the firmware everything is fine. After a few times initializing and deinitializing, I get error messages. 

I have a cheapo-version AI-Thinker ESP32-CAM installed this firmware like this: 

```
esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 micropython_camera_feeeb5ea3_esp32_idf4_4.bin 
```
When trying to initialize the camera in Thonny, I run:

```
import camera
camera.init(0, format=camera.JPEG, fb_location=camera.PSRAM)
```
which is fine for a couple of times, then results results in an error: 
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
OSError: Camera Init Failed
```
When running in a Terminal, the Error description is longer and gives more information:
```
MicroPython v1.18-74-gfeeeb5ea3 on 2022-02-02; ESP32-cam module (i2s) with ESP32
Type "help()" for more information.
>>> import camera
>>> camera.init(0, format=camera.JPEG, fb_location=camera.PSRAM)
E (17020) gpio: gpio_install_isr_service(449): GPIO isr service already installed
E (17020) camera: Camera probe failed with error 0x103(ESP_ERR_INVALID_STATE)
E (17030) camera: Camera Init Failed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Camera Init Failed
```
So I guess,  I will do more research to find out what is going wrong here.

SaitamaTechno, 2023-09-30:
Hello, if anyone still faces the issue, here is a solution:
You need to deinit the camera at the end of ever camera usage. Here is a sample code:
`import camera

camera.init(0, format=camera.JPEG)
#, fb_location=camera.PSRAM
camera.flip(1)
# left / right
camera.mirror(1)

# framesize
camera.framesize(camera.FRAME_QVGA)
# The options are the following:
# FRAME_240X240
# FRAME_96X96 FRAME_QQVGA FRAME_QCIF FRAME_HQVGA FRAME_240X240
# FRAME_QVGA FRAME_CIF FRAME_HVGA FRAME_VGA FRAME_SVGA
# FRAME_XGA FRAME_HD FRAME_SXGA FRAME_UXGA FRAME_FHD
# FRAME_P_HD FRAME_P_3MP FRAME_QXGA FRAME_QHD FRAME_WQXGA
# FRAME_P_FHD FRAME_QSXGA
# Check this link for more information: https://bit.ly/2YOzizz

# special effects
camera.speffect(camera.EFFECT_NONE)
# The options are the following:
# EFFECT_NONE (default) EFFECT_NEG EFFECT_BW EFFECT_RED EFFECT_GREEN EFFECT_BLUE EFFECT_RETRO

# white balance
camera.whitebalance(camera.WB_SUNNY)
# The options are the following:
# WB_NONE (default) WB_SUNNY WB_CLOUDY WB_OFFICE WB_HOME

# saturation
camera.saturation(0)
# -2,2 (default 0). -2 grayscale 

# brightness
camera.brightness(0)
# -2,2 (default 0). 2 brightness

# contrast
camera.contrast(0)
#-2,2 (default 0). 2 highcontrast

# quality

# 10-63 lower number means higher quality
def take_photo(qualityy):
    camera.quality(qualityy) #between 10-63 lower number means higher quality
    buf = camera.capture()
    #print(len(buf))
    return buf

def camera_deinit():
    camera.deinit()`