lemariva/micropython-camera-driver #40: Is it possible to remove the Micropython firmware?

pablojr, 2022-04-19:
Hi,

thank you for this nice project.
Not an issue per se, but is it possible to remove the Micropython firmware I've uploaded to a Esp32-Cam AI-Thinker? 
so to get back to upload C programs to the device?

Thanks.

leonardowolf, 2024-06-14:
search about esptool erase_flash for the right parameters, but as long as you use some ide like platformio or arduino, the memory from the espcam will be overlaped by the new firmware you are putting in it, so in other words just upload your code to the board and it will work

pablojr, 2024-06-19:
Yes, I guess I discovered that after some trial and error, and lot of Internet searching.
Indeed, the question should have been where to get the esp23-cam firmware to flash the device again, given this statement from AI Tinker "[The default firmware is already included in the factory, and no additional download is provided. Please be careful if you need to re-burn other firmware.](https://docs.ai-thinker.com/en/esp32-cam)"
However I think I have two ways of flashing the esp32-cam back: I'll buy a new device and I'll dump the firmware first thing on arrival. Or I can use any of the other firmware images built already; i.e. [esp32cam-rtsp](https://github.com/rzeldent/esp32cam-rtsp)