lemariva/micropython-camera-driver #44: RGB565 format is not working

e135193, 2022-07-02:
[state = camera.init(0, format=camera.RGB565, fb_location=camera.PSRAM)
print("Camera State: ",state)
camera.framesize(camera.FRAME_96X96)
buf = camera.capture()
buf
False](url)

AwesomeCronk, 2022-07-26:
Is this with the ESP32-CAM? I was able to capture with mine in RGB565 (latest commit as of 7-26-22). Grayscale was what I had problems with (I think it crashed the OV2640 itself, because I couldn't reinitialize the camera, even after rebooting the ESP32, until I power cycled everything for a few seconds.)

e135193, 2022-08-06:
Hello @AwesomeCronk I see that 2 Feb 2022 is the latest commit on this driver. "(latest commit as of 7-26-22)" Where can I find the commit that you mentioned ?

I have compiled micropython with this camera driver, but still I could not get it work on RGB565 format.

could you please share what you did step by step ?

AwesomeCronk, 2022-08-06:
@e135193 I'm sorry if I confused you! I hadn't checked the date of the commit, so I just said latest and put the date of writing. Unfortunately I don't see a newer commit either.