lemariva/micropython-camera-driver #48: F***k U

ABmamunHoneybee, 2022-09-01:
What problem u have!!!!!!!!!!, 
If you are not able to give us support, why u build this Firmware?

Every time, we are getting problems with the Init camera. Already people are posting this for help. But, u are sleeping about this.
We need a solution.

You have to solve this. ASAP

Lunatik00, 2022-09-02:
Understand what the license is:

   7. Disclaimer of Warranty. Unless required by applicable law or
      agreed to in writing, Licensor provides the Work (and each
      Contributor provides its Contributions) on an "AS IS" BASIS,
      WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or
      implied, including, without limitation, any warranties or conditions
      of TITLE, NON-INFRINGEMENT, MERCHANTABILITY, or FITNESS FOR A
      PARTICULAR PURPOSE. You are solely responsible for determining the
      appropriateness of using or redistributing the Work and assume any
      risks associated with Your exercise of permissions under this License.

This is the license, as you can read, it is an "AS IS" software, this applies to most open source software because support is not always viable, you are not entitled to insult people because you found a problem and they don't answer you in the time you want, you can always be the one that solves the issue and provides the solution to the community instead of insulting the people that do solve problems. Don't be entitled, it discourages real solutions because developers don't want to deal with people like the one portrait in your comment.

ABmamunHoneybee, 2022-09-07:
this firmware is totally garbage. I lost my 2-3 weak for this.
Nothing is working here, I try ADC and Camera. those both are not working. 

I don't think more about other modules. If u don't have the minimum ability to handle this kind of thing, don't give us the open source file. 

I recommend people don't use this firmware and don't waste time like me. Our time is not valueless like your time.

ABmamunHoneybee, 2022-09-08:
@Simpler1,
Do u have a solution?
If u don't, then don't give react.
I realize you have zero experience.

Simpler1, 2022-09-08:
Just ban this twat.  His abuse of the site is childish.

closed 2022-09-08 by lemariva