lemariva/micropython-camera-driver #73: Looping RTCWDT_RTC_RESET after flashing

jose1711, 2022-09-21:
After erasing memory and flashing the microcode (tried both version) I am getting:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

Any idea? Could it be hardware issue (board is from Aliexpress so quality is not guaranteed)? Thank you.

sjev, 2022-09-25:
I can confirm same issue :disappointed:  

How to reproduce: 

```
wget https://github.com/lemariva/micropython-camera-driver/blob/master/firmware/micropython_camera_feeeb5ea3_esp32_idf4_4.bin

esptool.py  --port /dev/ttyUSB0 --baud 460800 erase_flash
esptool.py  --port /dev/ttyUSB0 --baud 460800 write_flash -z 0x1000 micropython_camera_feeeb5ea3_esp32_idf4_4.bin
```

... the firmware itself seems to be a bit light,  a normal Micropython binary is around 1.5 Mb, this wone is just **144kB**

sjev, 2022-09-25:
check your binary. I've got the same error by using `wget` to get the firmware. It got html rubbish and saved it as `bin`.

jose1711, 2022-09-25:
> check your binary

Thank you, the corrupted (HTML output) was indeed the case (somehow it did not strike me that the upload was so superfast :facepalm:. Anyway even after getting the firmware correctly and multiple upload attempts it's showing the same. To share the full story:
- USB TTL is used to upload the code, logic is set to 5V, wiring is as follows (USB TTL -> ESP32CAM)
  - 5V -> 5V
  - GND -> GND
  - RX -> TX
  - TX -> RX
- on ESP32CAM IO0 is connected to GND
- both erasing and firmware upload report no issue
```
$ esptool.py  --port /dev/ttyUSB0 erase_flash --force
esptool.py v4.2.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ec:62:60:94:8f:84
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.4s
Hard resetting via RTS pin...

$ esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 micropython_cmake_9fef1c0bd_esp32_idf4.x_ble_camera.bin                 
esptool.py v4.2.1
Serial port /dev/ttyUSB0
Connecting...
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ec:62:60:94:8f:84
Stub is already running. No upload is necessary.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00181fff...
Compressed 1575808 bytes to 1010481...
Wrote 1575808 bytes (1010481 compressed) at 0x00001000 in 89.2 seconds (effective 141.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
- connection is tested with `picocom /dev/ttyUSB0 -b115200`
but this gives no response

- once IO0 to GND connection is removed `picocom` starts to show said loop:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

- if reset button is pressed during the connection the following message appears in the communication:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```