lemariva/micropython-camera-driver #75: Corrupted firmware?

sQur1, 2022-11-18:
Device: ESP32-CAM connected through the ESP32-CAM-MB's usb (MacOS)
Chip: ESP32-D0WD-V3
Software: Thonny
Bought on: Aliexpress

I've downloaded the 1.4MB firmware (https://github.com/lemariva/micropython-camera-driver/blob/master/firmware/micropython_camera_feeeb5ea3_esp32_idf4_4.bin) and flashed it and every time I do, it gives me an infinite loop. Corrupted files. I've tried flashing it from ESPtool in the terminal and also within Thonny, same result. I don't know what I'm doing wrong because the firmware completes flashing without errors. I've also tried the other firmware (https://github.com/lemariva/micropython-camera-driver/blob/master/firmware/micropython_cmake_9fef1c0bd_esp32_idf4.x_ble_camera.bin).

Here's the infinite looped message I get:

```
ELF file SHA256: ed0cb460b8dbabf0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
entry 0x40080618
E (536) psram: PSRAM ID read error: 0xffffffff
E (536) cpu_start: Failed to init external RAM!

abort() was called at PC 0x4008143c on core 0

Backtrace:0x40081e89:0x3ffe3bb00x4009061d:0x3ffe3bd0 0x400962dd:0x3ffe3bf0 0x4008143c:0x3ffe3c60 0x40079356:0x3ffe3c90  |<-CORRUPTED
```

The only firmware that works is the official one from micropython (https://micropython.org/download/esp32/). I've tried many other ones that I assume are all derived from Lemariva's firmware. None of them work either.

I'm really stuck here, I don't know what to do.

Is my chip not compatible?

algorni, 2022-11-22:
I've a similar issue, did you find an alternative?

algorni, 2022-11-22:
This is somehow related to this: https://github.com/lemariva/micropython-camera-driver/issues/73

algorni, 2022-11-22:
Did you flash from this address 0x1000?  with Circuit Python you should start from 0x0000   with uPython seems from 0x1000 it solved my issue!

sQur1, 2022-11-22:
> 

I flashed from 0x1000 with micropython using esptool, I flashed the default configs using Thonny (micropython) and the default settings in Arduino IDE... All of them gave me a PSRAM error. After days of researching, I believe my PSRAM is faulty. It's a clone from Aliexpress and apparently they're quite often faulty unfortunately. I can still use the board without PSRAM, but the whole point was to use it as a standalone timelapse camera. Oh well...

sQur1, 2022-11-22:
> Did you flash from this address 0x1000? with Circuit Python you should start from 0x0000 with uPython seems from 0x1000 it solved my issue!

I flashed from 0x1000 with micropython using esptool, I flashed the default configs using Thonny (micropython) and the default settings in Arduino IDE... All of them gave me a PSRAM error. After days of researching, I believe my PSRAM is faulty. It's a clone from Aliexpress and apparently they're quite often faulty unfortunately. I can still use the board without PSRAM, but the whole point was to use it as a standalone timelapse camera. Oh well...

closed 2022-11-22 by sQur1