lemariva/micropython-camera-driver #76: OSError: Camera Init Failed

wildernessfamily, 2022-12-15:
There have been many comments and posts relating to `OSError: Camera Init Failed` error when trying to initialize the camera. I have reviewed every post, which the most recent was over a half a year a go, and all have gone unanswered.
Steps that I have taken on my MacBookPro:
1. I have a 3 brand new AIThinker ESP32-CAM microcontrollers. 
2. Downloaded `micropython_camera_feeeb5ea3_esp32_idf4_4.bin` (commit it `905c798` on February 2, 2022)
3. Created a python virtual environment in a new folder (Python version 3.10.7).
```
mkdir -p sandboxESP32CAM
cd sandboxESP32CAM
python -m venv .venv
source .venv/bin/activate
```
4. Installed ESPTools `pip install esptool`
5. Performed the erase flash `python -m esptool --chip esp32 --port /dev/cu.usb1440 erase_flash`
6. Installed the firmware above. `python esptool --chip esp32 --port /dev/cu.usb1440 write_flash -z 0x1000 micropython_camera_feeeb5ea3_esp32_idf4_4.bin`
Everything was a success.
7. Opened VSCode `code .`
8. Installed PyMakr extension
9. Restarted VSCode
10. Created a PyMakr project
11. Copied over the code. I tried all three projects, webcam, webserver (both picoweb & MicroWebServer), and timelapse.
They all failed with the same error: `OSError: Camera Init Failed`
I tried with all three ESP32-CAM controllers.
I'm at a loss on what else I can try.

magixyu, 2023-03-03:
I have the same issue. Could you please help to take a look?

```
>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 11, in <module>
  File "webserver.py", line 48, in run
OSError: Camera Init Failed
```

PsuFan, 2023-07-12:
Make sure you execute your code after a hard reset and check to make sure the camera ribbon cable is fully inserted. I get this error only after a soft reset and am still looking for a solution. Works fine for me on first power up.

shillehbean, 2023-11-25:
I am also having this issue, so we have to hard reset every time we want to use the camera? Then that removes the use of practical use case unless I am missing something

PsuFan, 2023-11-25:
I was able to deinit before init to solve the issue on soft reset.

kdschlosser, 2024-11-20:
I know this is old but a soft reset only clears the RAM. It doesn't clear out any hardware settings. so if you do a soft reset and attempt to restart the camera it will fail. only a hard reset will clear the hardware states. 

The same problem exists with SPI as well.

PsuFan, 2024-11-20:
IMO init should do this automatically, not hope that the developer finds the issue and figures out how to fix it…

kdschlosser, 2024-11-20:
Micropyhton doesn't have anything build into it to let anything know that a soft reset is occurring. This is why most libraries do have a deinit function or are bound to `__del__` so when you delete the object it will do it's cleanup and release the hardware parts. The user knows when a soft reset is going to occur and is able to perform the cleanup that is needed bu using the deinit or deleting the object that holds any hardware resources.

I believe that there should be some way to register a callback function so that application or library C code can be notified and cleanup can be done that way. But that is not available so it's up to the user to do.

PsuFan, 2024-11-20:
Not sure what the implications of not cleaning up are or am really talking about that. Init can and should do the cleanup aka work every time.

kdschlosser, 2024-11-20:
doesn't work that way. the init method has no clue whether or not it should clean up after settings that don't exist anymore. Remember the memory has been cleared. How is it supposed to know what pins were used last time?

PsuFan, 2024-11-21:
It’s the only way I know of to get it to reliably start every time. Deinit doesn’t crash if run first at boot and presumably does nothing.

kdschlosser, 2024-11-21:
I can tell you that part of the problem is there is no "instance" or class for a camera that gets made. No way to to really keep track of it. I am working on writing the driver over again so it will compile properly with MicroPython 1.22 and to expose more of the esp32-camera feature set and hopefully correct the deinit issue.

PsuFan, 2024-11-21:
FYI I also have to take the picture twice and throw out the first one. The first picture is from BEFORE I requested it lol…