lemariva/micropython-camera-driver #77: Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

orhanemree, 2023-01-29:
Hi, I got this error while trying `camera.init`. It's like #35 but i couldn't figure it out. Any help?

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400df3e6  PS      : 0x00060730  A0      : 0x800d82a4  A1      : 0x3ffca3e0  
A2      : 0x3f425c58  A3      : 0x00000001  A4      : 0xffff8fff  A5      : 0x3ffca568
A6      : 0x00000000  A7      : 0x00000002  A8      : 0x800df3d2  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ffbf118  A13     : 0x3ffb1450
A14     : 0x3ffb144c  A15     : 0x00000028  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008e1ed  LEND    : 0x4008e1fd  LCOUNT  : 0xfffffff7

Backtrace:0x400df3e3:0x3ffca3e00x400d82a1:0x3ffca440 0x400d7a42:0x3ffca470 0x400d2450:0x3ffca4c0 0x400d24ae:0x3ffca560 0x400e20bd:0x3ffca600 0x400e8819:0x3ffca630 0x400e8951:0x3ffca650 0x40085ca5:0x3ffca670 0x400e205a:0x3ffca710 0x400e8819:0x3ffca740 0x400e8842:0x3ffca760 0x40118bdd:0x3ffca780 0x40118f44:0x3ffca810 0x400f89c8:0x3ffca850

ELF file SHA256: ed0cb460b8dbabf0

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4344
load:0x40078000,len:13816
load:0x40080400,len:3340
entry 0x40080618

closed 2023-01-30 by orhanemree

tobylai-toby, 2024-10-04:
sorry, i met the same problem, can you please tell me how did you solve that?

orhanemree, 2024-10-06:
Hi!
It's been a long time since I met this problem. It's interesting that library still has it. Unfortunately, I couldn't solve it either and gave up. I hope you find a way to solve it!