lemariva/micropython-camera-driver #80: After burning firmware problems

zxyup, 2023-04-06:
After_ burning firmware,it tells me ''Device is busy or does not respond. Your options:''   So how can i solve it

dvaneynde, 2023-04-21:
Similar issue here.
It does work with this driver: `micropython_cmake_9fef1c0bd_esp32_idf4.x_ble_camera.bin`. But then PSAM is unavailable, and other options aren't either. So I guess it is an older version?
But the camera module does not work with `micropython_camera_feeeb5ea3_esp32_idf4_4`.

randomsmith9, 2025-03-28:
> After_ burning firmware,it tells me ''Device is busy or does not respond. Your options:'' So how can i solve it

I know this is old, but in case it helps someone else: If you have the ESP32-CAM-MB board with reset button

Option A) Try using the arduino ide serial monitor and while connected press the reset button

Option B) Try using thonny serial monitor, but before launching it

      Open the Thonny configuration.ini file (Tools -> Open Thonny Data folder) and in [ESP32] section you need to add:
      
      dtr = False
      rts = False
      
      Worked for me!
      
      The longer story is here:  https://github.com/thonny/thonny/issues/1462
      
      From <https://forum.micropython.org/viewtopic.php?t=10151&start=10>