lemariva/micropython-camera-driver #82: how to support for ov5640?

guanjiawen008, 2023-05-05:
I have followed your steps to compile esp32, but I modified the file that supports mpconfig.h to support ov5640, and when I program successfully, I still can't import the camera module, why? I don't have any errors when I look at the compilation, can God help compile a firmware that supports ov5640?

garvitkhera, 2023-08-15:
Hi @guanjiawen008, Did you find a solution to this? I'm also trying to run an OV5640 camera with the ESP32 using micropython. Do you have any suggestions or tips, please? TIA.

guanjiawen008, 2023-11-21:
I just have a complie fireware,if you want the fireware, i can email to you

garvitkhera, 2023-11-21:
Hi @guanjiawen008 

Thanks so much for your reply.

After I couldn't find much support for the same, I went ahead with C++.

But yeah, if I would again hop onto micropython for something similar, I would love to check out your firmware code. That'd be great if you could share that via email to garvit185@gmail.com.

Cheers,
Garvit