lemariva/micropython-camera-driver #105: Expose grab_mode config

FredrikMorstad, 2025-05-02:
I have a problem where the camera images lags one picture behind, similar to #95. The esp32-cam is running a server with an endpoint allowing for fetching the current view from the camera
```python
@app.route('/picture')
async def picture(request):
    # takes two images as the first one is from the previous capture
    # hack until grab_policy gets exposed in the driver
    camera.capture()
    buff = camera.capture()
    return buff, 200, {'Content-Type': 'image/jpeg', 'Content-Length': len(buff)}
```
As of now I have to capture the two images to get the current view of the camera. This would be solved if the micropython driver exposes the esp32-cams grab_mode so one can pass `CAMERA_GRAB_LATEST`

```
    camera.init(0, format=camera.JPEG, fb_location=camera.PSRAM, grab_mode=camera.mode.latest)
```
I can look into adding this, as it should be a simple update here
https://github.com/lemariva/micropython-camera-driver/blob/1392e59751a5140c7915dd8965705301e8640160/src/modcamera.c#L165

cnadler86, 2025-05-02:
New driver: https://github.com/cnadler86/micropython-camera-API

This exposes the grab mode

FredrikMorstad, 2025-05-02:
awesome, this is exactly what I need, thanks!

closed 2025-05-02 by FredrikMorstad